autonomousvision/kitti360Scripts #1: No fisheye image semantic label?

happyboyneu, 2020-10-21:
Hello, I have download this dataset,and I found that the fisheye images don't have corresponding semantic label. When do you release the fisheye semantic labels?

yiyiliao, 2020-10-21:
Hi, we do not have per-pixel semantic/instance labels for fisheye images. We will instead provide projected 3D bounding boxes for fisheye images by December.

happyboyneu, 2020-10-21:
What is "projected 3D bounding boxes"? Can it be used to do 2d detect?

yiyiliao, 2020-10-21:
You can find examples of "projected 3D bounding boxes" on perspective images here: https://www.youtube.com/watch?v=OonvYU5bx3s (at 2:04). We will provide similar labels for the fisheye images. Naively converting these to 2D bounding boxes might result in oversized 2D boxes.

happyboyneu, 2020-12-22:
Hello, It's already December, when will you release the "projected 3D bounding boxes" for fisheye image?

zmh20062257, 2021-12-09:
Hello,  you has been released the "projected 3D bounding boxes" for fisheye image?