autonomousvision/kitti360Scripts #3: Clarification for semantics of the pose data and 3d boxes -- what is the world frame?

pwais, 2020-10-25:
I'm having trouble plotting the 3d boxes against the velodyne clouds and it occurred to me that the pose and 3d box data isn't really documented yet on the website ( http://www.cvlibs.net/datasets/kitti-360/documentation.php ).  Could you please clarify the following?
 * All of the polygons in the `data_3d_bboxes` XML files appear to be cuboids.  While all cuboids have the same height, length, and width (i..e vertices  are all +/- 0.5m), the RT transform appears to scale the cuboid accordingly.  But _what coordinates are the transformed points in_ ?  What are the x, y, z axes of the points?  Are they in the world frame?
  * The velodyne frame is forward = +x, left = +y, up = +z.  What are the axes of the world frame?  It looks like perhaps forward = -y, left = x, up = z, but I'm not sure this is correct.  
  * What is `data_poses` / `cam0_to_world` ?  Is this the pose of the left camera frame in the world frame?  Does this transform include some change of axes from world to camera?  I.e. for the camera, +z is forward, but for the world frame it looks like +z is up.
  * what is `data_poses` / `poses.txt` ?  These matrices appear to have a different shape and aren't used anywhere.
  * I tried mapping the `data_3d_bboxes` points to the velodyne frame using `cam0_to_world`, `calib_cam_to_velo`, and the RT for the individual box.  Is this correct?  In the old KITTI READMEs, there was an explicit formula written to make the math clear (because sometimes the names were confusing; e.g. `calib_cam_to_velo` actually looks like cam _from_ velo).  Could you please perhaps write out the expected chain of transforms given the data available?
  * Why are some poses missing in `data_poses` ? (i.e. for some frames there is no pose entry).  Would it be fair to linearlly interpolate translation and slerp rotation for missing frames?
  * In the original KITTI dataset, there is a non-zero rotation in the oxt to velodyne frame transform. (I.e., oxt and velodyne use different axes, but there's also a little yaw/roll/pitch change between the two based upon calibration).  What is the world frame in KITTI-360?  Is it the oxt frame, or some other frame?

yiyiliao, 2020-10-26:
Thanks for the feedback! I have updated the documentation page regarding calibrations, poses and bounding boxes. Please check that out first and I will address the rest of your questions asap.

pwais, 2020-10-26:
Hi @yiyiliao , thanks for the quick reply!  I still have a few questions though:
 * What is the center of the "system frame" and what are the axes? Is it the car's rear axel?  Are the axes +x = forward, +y = left, +z = up as in the velodyne frame?
 * `data_3d_semantics` -- 
     * "The vertices and faces matrices form the mesh of the bounding box in a local coordinate" -- What are the axes of the object frame?  Is it +z = up, +x = forward, +y = left like the velodyne frame?  
     * Are the vertices always in the same order?  (E.g. are the first four always the front face?)  
     * "The transform matrix transforms this mesh to the world coordinate" -- is this the same world frame as in the static and dynamic fused lidar clouds ?
 * `calibration/calib_cam_to_velo.txt`:  I believe this is actually camera _from_ velodyne and not camera _to_ velodyne.  When I project velodyne points into the camera frame, I have to take the inverse of this transform.
 * `data_poses`:
    * `poses.txt` - "Each line ... is a 3x4 matrix denoting the system pose in a global Euclidean space."  Is "global Euclidean space" the "world frame" mentioned elsewhere in the documentation?  Are these poses from GPS / IMU data or are they deduced from the lidar fusion process?  
   * `cam0_to_world.txt` -- "Each line .. is a 4x4 matrix denoting the pose of camera 0 in a global Euclidean space."  Does this transform include the change-of-axes between the world frame and the camera frame?  E.g. if the world frame has +x = forward, but the camera has +z = forward, then does the transform's rotation include this change?  Also, the code appears to take the inverse of this transform, so perhaps the name is backwards?
 * `calibration/calib_cam_to_pose.txt` -- "Each line contains a 3x4 matrix denoting the transformation from a camera to the system pose"  -- don't you mean the frame of the left camera and not "system pose" ?  
 * Did the KITTI-360 files recently get changed on the server?  I didn't get the fisheye yaml calibration files nor the point cloud `timestamps.txt` files in what I downloaded.  
 * Will the timetsamps for the cameras be made available?
 * For frames that do not have poses in `cam0_to_world.txt`, would it be fair to interpolate a pose based upon the nearest frames?  (e.g linearly interpolate T and slerp R ?)

pwais, 2020-10-27:
Looking at `T_left` of the "cam0 to ego" transform listed in `calib_cam_to_pose.txt`:
```
[[ 1.5752681],
 [ 0.00439141],
 [-0.65]]
```
and comparing it with the car diagram [image]

`T_z` suggests that the ego frame might be the IMU, since the 0.65m number matches there, but the `T_x` and `T_y` look either incorrect or the picture is rounded.  Moreover, `T_right` of the right camera isn't as precisely consistent with left camera, but perhaps that's because the left camera is tilted a lot.

So I'm not sure about the calibration data provided.  It would really help to have the "system frame" and "world frame" documented-- in particular, the semantics of the axes.  It looks like the fused lidar clouds obey these axes and not the velodyne axes.

Inspecting a single cuboid label, I see this:
```
obj['cuboid'] - np.mean(obj['cuboid'], axis=0
array([[ 0.60811761,  0.40435617,  0.59196624],
       [ 0.60371577,  0.3976227 , -0.60096458],
       [ 0.50124741, -0.52474226,  0.59760486],
       [ 0.49684557, -0.53147572, -0.59532596],

       [-0.50124741,  0.52474226, -0.59760486],
       [-0.49684557,  0.53147572,  0.59532596],
       [-0.60811761, -0.40435617, -0.59196624],
       [-0.60371577, -0.3976227 ,  0.60096458]])
```

suggesting the cuboid is in this order:
```
+x +y +z
+x +y -z
+x -y +z
+x -y -z

-x +y -z
-x +y +z
-x -y -z
-x -y +z
```

That order looks a bit odd-- the front (+x) and rear (-x) faces wind in the same way about +x, but points are out of order-- and it's notable that the [KITTI-360 code](https://github.com/autonomousvision/kitti360Scripts/blob/fc4e92bfe7d7da0a404e58bca3b98660147ca09c/kitti360scripts/helpers/project.py#L17) appears to flip the y and z axes, *but only for the rotation* component of the object's transform :(   It's not really clear if `faces` is suppose to mean some permutation of these vertices, since `faces` has 12 rows.  When I plot all the cuboids with a fixed rotation (`I`), it does appear that there's some disagreement in which way is positive yaw in the object frame versus in the ego frame:

[image]

However, this slant looks larger than 5 degrees, and the velodyne and IMU appear to be on-axis... the left camera is only tilted 5 degrees ...

yiyiliao, 2020-10-28:
Yes, the ego-frame is the IMU. The "world frame" is converted from IMU measurements using the Mercator projection. You can find more details about this projection in this [paper](http://www.cvlibs.net/publications/Geiger2013IJRR.pdf). You are right that the fused lidar clouds obey axes of this world frame, as you can find out in the [accumulation code](https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/devkits/accumuLaser/src/accumulation.cpp#L465).

The original coordinate system of the IMU, as shown in the car diagram, is defined as "x = forward, y = right, z = down". We converted it to "x = forward, y = left, z = up" for convenience.

I am not sure if I understood your question regarding the 3D bounding box correctly. This [code](https://github.com/autonomousvision/kitti360Scripts/blob/fc4e92bfe7d7da0a404e58bca3b98660147ca09c/kitti360scripts/helpers/project.py#L17) does not flip any axis. It applies a rotation to every point instead. The transform matrix transforms the bounding box to the "world frame" on which the fused point cloud is defined.

pwais, 2020-10-30:
@yiyiliao Thanks for the added details!  Ah, I should have consulted the original paper.  (It might be helpful to link to it on the documentation page-- yes the paper does have some key context).

> This [code](https://github.com/autonomousvision/kitti360Scripts/blob/fc4e92bfe7d7da0a404e58bca3b98660147ca09c/kitti360scripts/helpers/project.py#L17) does not flip any axis

Hmm, I read `np.matmul(R, points.transpose(0,2,1)).transpose(0,2,1) + T` as flipping y and z, but only for application of the rotation matrix.  (To be honest though, I do find this code very hard to follow).  

Now that I understand the world frame better, let me see if I can fix my plots.

@yiyiliao -- One outstanding question: thank you for making the velodyne and SICK timestamps available, but do you think you might make the camera timestamps available too?  In the original KITTI Tracking & Object Benchmarks, [I observed there's about a 10ms difference between velo and camera](https://github.com/pwais/psegs-kitti-ext#sensor-sample-rates-are-consistently-10hz) (which is very nice!).  Would love to verify that for KITTI-360.  Moreover, the timestamps would perhaps be a bit more relevant for the side (fisheye) cameras in KITTI-360, especially since it looks like the intention is to use lidar labels for those cameras.  (Finally, the process of linking the timestamps in the old KITTI data was a bit tedious, as seen in my repo there, and it would be very cool for KITTI-360 to simply have all timestamps available).

pwais, 2020-11-02:
Ok @yiyiliao thank you I think I figured things out!  I would recommend updating the documentation ( http://www.cvlibs.net/datasets/kitti-360/documentation.php ) to (1) note that "system pose" means "IMU pose" and (2) note the world coordinate frame axes as you describe here.  It would also be helpful to explain what the "faces" data means for the cuboid labels... as far as I can tell that data can be ignored?

Some of the orientation of the cuboid labels looks odd (in these plots, the lighter colors mean front of object), but otherwise the labels look good (!)

[image]

[image]

[image]

yiyiliao, 2020-11-04:
Great to hear that you figured things out @pwais and thanks a lot for your suggestions for improving the documentation! Sure, I will adapt the documentation accordingly.

The faces combined with the vertices represent a triangular mesh of the bounding box. They are not needed if you draw the wireframes only.

pwais, 2020-11-04:
Thank you! Going to close this issue-- will file any other questions elsewhere.

closed 2020-11-04 by pwais