autonomousvision/kitti360Scripts #6: Meshes in bounding box labels consist of non-sense numbers

cmpute, 2020-11-16:
Hi, thank you for publishing this amazing dataset!

I downloaded the bounding boxes labels just to find that the meshes defined in the xmls are all invalid. They're all like the following
```xml
  <vertices type_id="opencv-matrix">
    <rows>8</rows>
    <cols>3</cols>
    <dt>f</dt>
    <data>
      5.00000000e-01 5.00000000e-01 5.00000000e-01 5.00000000e-01
      5.00000000e-01 -5.00000000e-01 5.00000000e-01 -5.00000000e-01
      5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
      -5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
      5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
      -5.00000000e-01 -5.00000000e-01 -5.00000000e-01 5.00000000e-01</data></vertices>
  <faces type_id="opencv-matrix">
    <rows>12</rows>
    <cols>3</cols>
    <dt>u</dt>
    <data>
      0 2 1 2 3 1 4 6 5 6 7 5 4 5 1 5 0 1 7 6 2 6 3 2 5 7 0 7 2 0 1 3 4
      3 6 4</data></faces>
```

It's just a trivial unit cube. Is this intended or actually bug in the label?

pwais, 2020-11-17:
@cmpute  personally I still don't understand `faces`, but the cuboids are there.  meshes maybe come in a later release?

yes, the `data` thing is kind of garbage-- every cuboid is a 1-meter cube centered at the object origin.  the actual object length, width, and height are encoded as scaling coefficients into the object's RT.  Here's how I deduced the cuboid in the world frame, and note citation to code in this repo for another example: https://github.com/pwais/psegs/blob/4b53c62c12b19d26af757bd2eeb3be0d199294e1/psegs/datasets/kitti_360.py#L243 

If you want to extract the actual length, height, and width (the scale) from the object transform RT matrix, you have to do some math: https://github.com/pwais/psegs/blob/4b53c62c12b19d26af757bd2eeb3be0d199294e1/psegs/datasets/kitti_360.py#L786

Hope this helps!

cmpute, 2020-11-17:
@pwais Oh that's really interesting, I always assume that the R|T matrix will be a rigid transformation... Really thanks for pointing this out, I will take a look at the data.

cmpute, 2020-11-17:
@pwais by the way, the faces are the indices of vertices that are used to construct a mesh. It seems that the "bounding boxes" are not all boxes, but also some other shapes. So in that case, the faces will be used.

pwais, 2020-11-18:
@cmpute uggg which sequence did you find non-box objects? I’ve plotted several and so far I only see cuboids.  I thought the mesh thing was because they were going to add non-cuboid labels later.

cmpute, 2020-11-18:
@pwais Ah I didn't find one actually, but I found there are sphere-like objects in the demo video. Maybe they will be released later.

pwais, 2020-11-18:
@cmpute Thanks for clarifying!  Do post if you find odd stuff!  The old KITTI data had some odd problems like a few missing velodyne scans and the odometer benchmark had some test / train overlap. But for KITTI-360 we have this github repo so it’s easier to share.

They already updated a couple of the datasets to add timestamps for camera frames and point clouds.  I think they’re going to release the mesh things like the trees / vegetation in the video some time in ... January?  I am not part of the project but I believe they want to use the mesh / volume labels and fused lidar clouds to project to the fisheye cameras and impute labels for those cameras.  Note that the intrinsics of the fisheye cameras aren’t available yet— there’s definitely going to be a future release for that stuff.

Also note the pose data (world to ego) is also incomplete.  Not sure if they’re going to backfill or if it’s intentionally missing (e.g. for some future odometry competition).

yiyiliao, 2020-11-18:
Hi @cmpute , thank you for the question! The bounding boxes can be transformed into the world coordinate using the transformation matrix in each object, e.g.,

```
  <transform type_id="opencv-matrix">
    <rows>4</rows>
    <cols>4</cols>
    <dt>f</dt>
    <data>
      2.92481422e+00 -1.63844514e+00 2.43007307e-04 9.66490295e+02
      4.16620302e+00 1.15024292e+00 -7.32338813e-04 3.78474097e+03
      1.53910322e-03 8.15824198e-04 1.52057445e+00 1.14976791e+02 0. 0.
      0. 1.</data></transform>
```

which includes scaling, rotation, and translation.

cmpute, 2020-11-19:
@yiyiliao @pwais Thanks for your help! I managed to plot the bounding boxes now :+1:

closed 2020-11-19 by cmpute