autonomousvision/kitti360Scripts #7: Not Found for url: https://pypi.org/simple/kitti360scripts/

lym-ay, 2020-11-24:
when i run ''python -m pip install kitti360scripts', console show "requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://pypi.org/simple/kitti360scripts/".I find kitti360scripts from network of https://pypi.org,but I don't find it. what I do it?

LuwooChen, 2020-11-26:
I got the same problem, is there anybody who can help?
Appreciate.

yiyiliao, 2020-11-26:
Sorry for the late reply. This should work:
'''
pip install git+https://github.com/autonomousvision/kitti360Scripts.git
'''

lym-ay, 2020-11-30:
> Sorry for the late reply. This should work:
> 
> ```
> pip install git+https://github.com/autonomousvision/kitti360Scripts.git
> ```

thank you ~

lym-ay, 2020-11-30:
I use the code from 

> I got the same problem, is there anybody who can help?
> Appreciate.

I  don't download it using command line "python -m pip install kitti360scripts". I use the code from git and so I can run .