autonomousvision/kitti360Scripts #11: How to get labeled, individual point clouds?

Pyrestone, 2020-12-16:
Hi, I would like to have a set of point clouds in the vehicle's **local coordinate frame**, at each timestamp, with corresponding semantic and bounding box labels.

Basically, the **instantaneous observations** of the LIDAR sensor, labeled with semantic and bounding box labels, in a local coordinate system.

How would I go about getting that?

The labels are all in the world's coordinate frame, and don't seem have timestamps to discern which frame they came from.  
Additionally, the binary lidar frames don't seem to have any world coordinate association.  
I'm sure you have this information available, since you must have somehow done the inverse thing already.

Please provide me with some hints on how I could accomplish this. The documentation is not very clear on how to get this information.

pwais, 2020-12-28:
@Pyrestone Here's a draft of code I have used to get cuboid labels in the ego frame: https://github.com/pwais/psegs/blob/ee6d7260ca604a0d121d24c1a4f556b1618e7f2b/psegs/datasets/kitti_360.py#L767   I've tested and the cuboids reproject into the camera OK.  

The code I used to parse the cuboid XML labels is pretty similar to that in this repo:   https://github.com/pwais/psegs/blob/ee6d7260ca604a0d121d24c1a4f556b1618e7f2b/psegs/datasets/kitti_360.py#L243

The world-to-ego transform is in the `poses.txt` file.  Note that some poses are NOT available!  For each sequence, there are some stretches of segments both with and without pose data.  Be careful to program defensively.

Hope this might help!

leezonpen, 2020-12-30:
@pwais Hi, when i parse the point clouds and labels, i find that some bbox were dispeared. For example in the sequence 2013_05_28_drive_0000_sync frame id 386, the bboxs behind the car are dispeared but they are still exist in the last frame. 

And i am also confused about that there is no bbox before frame id 4300+ in the sequence 2013_05_28_drive_0002_sync. 

Do you have the similar problem about the data?

Pyrestone, 2020-12-30:
@leezonpen I found the same issue as well. Checking into the data, I found that drive_0001 has no 3d semantic labels at all, and drive_0002 also only has them from 4300 onwards. You can see this directly in the data_3d_semantics filenames.

pwais, 2020-12-31:
drive 0000 has labels and ego poses for several sub-segments in case it helps.  the labels are useless without the ego poses even if present

leezonpen, 2020-12-31:
I used poses.txt to make sure which frame was availabel, but i still found some objects had no bboxs in the annotated frame. For another example, a moving car which is near ego vehicle is without bbox in the drive_0000 frame 14. I checked there is no dynamic object label in the xml file for this frame.

yiyiliao, 2021-01-15:
Hi @Pyrestone , we released the [accumuLaser](https://github.com/autonomousvision/kitti360Scripts/tree/master/kitti360scripts/devkits/accumuLaser) which we use to convert point clouds to the world coordinate and to accumulate them. You can use this to find the correspondence between the individual point cloud and the labeled point cloud (e.g., for each point in the individual point cloud find its nearest neighbor in the labeled point cloud). Note that we cropped the accumulated point cloud such that a point is filtered if its distance to the closest camera is larger than 30 meters. Thus a distance threshold should be considered when you want to retrieve labels of the individual point cloud.

yiyiliao, 2021-01-15:
Hi @leezonpen , we did not release any labels before frame 4300 of drive_0002 as they will be used as a part of our held-out test set. We have discarded some driving sequences (e.g. drive_0001) and thus the ID is not guaranteed to be continuous. 

Regarding the missing of a bbox at a specific frame, do you mean 2D or 3D bbox? If you mean 2D bbox extracted from the 2D instance label, this could happen occasionally due to inconsistent 2D labels, especially for small and far objects. The labels should be consistent in most of the cases though. However, if you project a 3D bbox directly to each 2D frame (like https://youtu.be/OonvYU5bx3s?t=94), I don't see why that the bbox would disappear.

JulesSanchez, 2021-02-18:
Hi @Pyrestone, I rewrote a part of accumuLaser in python so I could use it more easily (I didn't write what was relative to SICK data, and the sparsify function as it was meant to recover original frame and not reproduce the accumulated pointcloud), and I added some functions to recover labels for individual point cloud based on a nearest neighbor search in the accumulated pointclouds.
You can find this work at https://github.com/JulesSanchez/recoverKITTI360label