autonomousvision/kitti360Scripts #14: Label discrepancy

pimonteiro, 2021-01-25:
Hello! Before anything, what a great work you did!
My question: upon analysing the labels from the segmentation mask, i noticed that more labels exists than the cityscapes. Being more specific, i noticed there are extra classes, from 34 to 39. Can you explain me why do these exist while you specified making the dataset in accordance to the cityscapes format?

Thank you!

pimonteiro, 2021-01-25:
[image]

For example, on the image above, there is a class 35 on the red portion of the image (I assume it points to some sort of gate, just behind the car).

pimonteiro, 2021-01-26:
I apologize for my lack of understanding of the paper and the code provided in this github. After carefully analysing it I finally understood the label system.

closed 2021-01-26 by pimonteiro

yiyiliao, 2021-01-26:
Hi @pimonteiro, thanks for your question! We indeed have more classes than Cityscapes as we annotated in that way, i.e., we have a few more classes such as "garage". We plan to use the same set of labels for training and evaluations though. Great that you figured it out already, feel free to let me know if there is anything we should improve regarding the paper or the code.