autonomousvision/kitti360Scripts #15: Fisheye Camera Velodyne Pose in calib_cam_to_velo.txt?

wsolichin-sees, 2021-05-08:
First of all, thank you for providing the KITTI-360 dataset. 

Quick question with the fisheye camera position, will it be possible to provide the fisheye camera poses with respect to the Velodyne LiDAR be provided in the calib_cam_to_velo.txt? Or do we have to refer to the sensor locations diagram and get the pose?

Thank you for your help.

yiyiliao, 2021-05-09:
Thank you for looking into our dataset!

We do not have extra extrinsic parameters for the relative transformation between fisheye cameras and the Velodyne. You can retrieve them via cam02/03 -> cam00 -> velo, where cam00 -> velo is provided in calib_cam_to_velo.txt and cam02/03 -> cam00 can be calculated from calib_cam_to_pose.txt.

wsolichin-sees, 2021-05-10:
Thank you for your info!

closed 2021-05-10 by wsolichin-sees