autonomousvision/kitti360Scripts #19: Dependencies missing in setup.py

samschulter, 2021-05-14:
Hi.

I installed this package via `pip install git+https://github.com/autonomousvision/kitti360Scripts.git` under `miniconda` and Python 3.8. But when I try to do imports, some packages are missing, for instance:

```
from kitti360scripts.helpers.project import CameraPerspective as Camera
> ModuleNotFoundError: No module named 'yaml'
from kitti360scripts.helpers.annotation  import Annotation2D
> ModuleNotFoundError: No module named 'skimage'
```

I think the packages `pyaml` and `scikit-image` should be added in `setup.py` under `install_requires`.

yiyiliao, 2021-05-15:
Thanks for reporting the error! It should be fixed now.

samschulter, 2021-05-18:
Thanks for the quick update!! Those two things seem to be resolved, but I still get an error for some imports when installed via `pip install git+https://github.com/autonomousvision/kitti360Scripts.git` into an empty repository.

Specifically, `from kitti360scripts.helpers.project import CameraPerspective as Camera` gives the error "ModuleNotFoundError: No module named 'kitti360scripts.devkits'". I think the reason is that the directories "devkits" and "devkits/common" are not part of the library. So, when installing directly from git, these directories are not even downloaded. **I think (but not sure)** that's because the directories "devkits" and "devkits/common" do not have `__init__.py` files and are not found by "find_packages" in `setup.py`. So you could move this function into `helpers` or the install instructions just say that the repository needs to be cloned first and then locally installed (this way all the files are available).