autonomousvision/kitti360Scripts #20: List of labels for bounding boxe annotations

nlgranger, 2021-05-21:
The labels for the bounding boxes annotations do not seem to match the list in https://github.com/autonomousvision/kitti360Scripts/blob/b74933bb4d358d4b8f4c371528f036be2092060c/kitti360scripts/helpers/labels.py which is for semantic annotations For instance, this list does not mention "Big Pole"

yiyiliao, 2021-06-09:
Hi @nlgranger, thanks for the feedback! It is true that the label names of a few classes are slightly inconsistent but the indices should match. For example, the "semanticId" of "bigPole" in the bounding box annotations (*.xml) is 21, which is referred to "pole" according to its "kittiId" https://github.com/autonomousvision/kitti360Scripts/blob/b74933bb4d358d4b8f4c371528f036be2092060c/kitti360scripts/helpers/labels.py#L83.
Sorry for the inconsistency. We will improve this in our future release.

nlgranger, 2021-06-10:
All right, thanks for the clarification.

closed 2021-06-10 by nlgranger