autonomousvision/kitti360Scripts #23: Split of PLY files

resha1417, 2021-06-27:
Hello team,

Thank you so much for wonderful work and preparing a large dataset.
Actually I am working on this dataset. As you mentioned in http://www.cvlibs.net/datasets/kitti-360/ in documentation part(3D Data Format) that data_3d_semantic consists of pointclouds which divided the fused point clouds into windows for example 000002_000385(start-frame_end-frame). 

I am using this dataset for 3D semnatic Segmentation where I will need x, y,z and semantic_id as an input to my network. I am not sure about data_3d_raw which has bin files corresponding to the same frame in ply file as well. For example 0000000002.bin which contains x,y,z is corresponding to the first row of 000002_000385.ply. As I want semantic label for 0000000002.bin file. 

For now I am just using ply file,  reading x,y,z and semantic label from that ply file only. But as it contains too many frames data ,  my gpu(32 gb capcacity,with batch size =1) is not able to handle such large input. I want to divide those ply files per frame. Can you please let me know how it is possbile?  Also, Can you please let me know how can I get semantic_id corresponding  to .bin file?

It would be great help.

Thanks and Regards,
Resha Thacker

yiyiliao, 2021-07-05:
Hi Resha, thanks for using our dataset and sorry for the late reponse. What do you mean by diving ply files per "frame"? It is true that each of our ply file is fused over multiple frames, but I am not sure if this is what you mean.

resha1417, 2021-07-05:
Hello @yiyiliao , Yes. I intend to use the data from ply files for each frame and not the multiple frames. Can you please tell me how shall i divide the ply file to read a single frame data? The other solution could be to provide 'semantic_ids' corresponding to the '.bin' files.

resha1417, 2021-07-06:
Hello @yiyiliao ,

Thank you for the reply. I used the original point cloud files(.ply files), however, it gives memory issue while training my network. I keep the batch size of 1 and use 32GB Tesla V100 GPU. Do you have any ideas on this? 

For now, i divided each ply file into 50 frames by this logic: 'length of file/50' . Using this logic I am able to train my network, however is the logic used by me is correct? 

Or do you have any idea how can i train my network without GPU memory error. Please let me know.