autonomousvision/kitti360Scripts #27: code for stitching the two fisheyes to make an equirectangular image

haruishi43, 2021-08-30:
Thanks for the great dataset!

I know it wasn't this dataset's intentions, but I was wonder if it is possible to make a 360 image (equirectangular image) from the two 180 fisheye cameras.
If someone did it or if you know any good method to go about this, it would be helpful.

I tried to go do it naively and made a dual-fisheye image and tried to stitch it on the Theta camera's conversion app, but it didn't work out. I haven't really done fisheye conversions, but my guess is that the two fisheyes are too far apart to create a perfect equirectangular image.
Then again, it seems like the 180 fisheye seems to be capturing more than 180 degrees (more like 190 degree?) 

Any help would be appreciated.

helmifraser, 2021-09-04:
Hi, I've tried tackling this problem myself to limited success.

Going by [this](https://stackoverflow.com/a/58336312) answer on Stack Overflow, newer versions of FFmpeg have a [v360 filter](https://ffmpeg.org/ffmpeg-filters.html#v360) to convert fisheye/dual fisheye images to equirectangular, which can take either a single image or a video. So I concatenated a fisheye pair to get a dual fisheye image, tested some FOV parameters of the filter (trial and error, really) to get this:
[image]

Obviously, the sharp border isn't great, which led me down another rabbit hole about blending and saw [this open enhancement ticket](https://trac.ffmpeg.org/ticket/8459). Trying a couple of the suggestions in there from Michael Koch (for windows batch, but the arguments for bash are similar enough) got me this:
[image]

Still isn't ideal, but slightly better than before. I'm sure someone more familiar with blending/stitching can do a much better job than I did. Michael Koch also links his book with some more examples on the filter: http://www.astro-electronic.de/FFmpeg_Book.pdf.

haruishi43, 2021-09-05:
@helmifraser Thanks for sharing with me your workflow.
Wow, that looks so much better than mine.

I tried using Hugin to stitch, but can't get reliable control points since it doesn't overlap as much.
[image]

I thought it would be easy, but I too fell in the rabbit hole :(

whu-lyh, 2023-03-21:
@helmifraser hi helmifraser and @helmifraser helmifraser, do you guys know how to convert the fisheye image to an equirectangular image in a batch?

closed 2023-07-17 by haruishi43

cuixing158, 2024-10-28:
@whu-lyh you can reference here: https://github.com/cuixing158/hugin-panorama