autonomousvision/kitti360Scripts #30: Can you release the datalodaer of votenet? I donot know how to load the dataset,If you can release the code?

czy-0326, 2021-10-03:


czy-0326, 2021-10-04:
Can you tell me How can I projrct the 3d bounding boxes to velo_points?

yiyiliao, 2021-10-04:
Thank you for your questions. We will release the dataloader of the baselines after cleaning up the code.

For projecting 3D bounding boxes to velo_points, the 3D bounding boxes are provided in the world coordinate. You can check this [accumulation code](https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/devkits/accumuLaser/src/accumulation.cpp#L450-L465) to see how the velodyne points are projected to the world coordinated and fused. You need to invert this process to project the 3D bounding boxes from the world coordinate to the local velodyne coordinate. We will improve our documentation and development toolkits to make this clearer.