autonomousvision/kitti360Scripts #37: About inverse function of img2cam for fisheye model

qianyu220, 2021-11-01:
I see the function cam2img to project 3d point to fisheye image, but do you have an inverse function of img2cam for your fisheye model?

yiyiliao, 2021-11-02:
Sorry we do not have it yet. The unprojection function is not as straightforward as the perspective camera due to the undistortion operation:
https://github.com/autonomousvision/kitti360Scripts/blob/8db21c6966dcfbffa2a9eec875e2ac0afa78f0ad/kitti360scripts/helpers/project.py#L180-L182

qianyu220, 2021-11-02:
Thanks for your reply. Looking forward to your update if possible.

eslambakr, 2022-04-27:
Dear @yiyiliao and @qianyu220 
Do u have any updates regarding this?

yuliangguo, 2023-08-08:
> Sorry we do not have it yet. The unprojection function is not as straightforward as the perspective camera due to the undistortion operation:
> 
> https://github.com/autonomousvision/kitti360Scripts/blob/8db21c6966dcfbffa2a9eec875e2ac0afa78f0ad/kitti360scripts/helpers/project.py#L180-L182

Does this mean a pixel-wise mapping recording the polinomial solution is needed for the undistortion?  It will be very helpful that the authors could help general a official version so that everyone can use the correct one.