autonomousvision/kitti360Scripts #38: What is K for fisheye cam?

Alvin0629, 2021-11-01:
Hi, 

if I want to use cv2.omnidir() function for fisheye undistortion, we need to know K, D and Xi. I see D and Xi in image_02.ymal, but where is the fx and fy in this case? Thanks for your reply!

A.L.

yiyiliao, 2021-11-02:
Hi,

I am not familiar with cv2.ominidir() function. Comparing the projectPoints() function of cv2.ominidir():
https://github.com/opencv/opencv_contrib/blob/master/modules/ccalib/src/omnidir.cpp#L143-L165
to our projection function:
https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/helpers/project.py#L174-L198
I think `gamma1` and `gamma2` in our code correspond to `f[0]` and `f[1]` in their implementation.

Btw, it is important to make sure that both adopt the same camera model, which is true as cv2.ominidir() also adopts the MEI camera model. 

Best,
Yiyi

tuanho27, 2022-02-15:
Hi @Alvin0629 ? Have you finished testing the undistortion for fisheye image yet? Could you share the example result?

Thanks & BRs,
TuanHo

tuanho27, 2022-02-17:
#Update
I tested the cv2.ominidir() to undistort the image using all given parameters (gamma, Xi, D, u,v), but the rectify output seems stretching at the edge, @yiyiliao do you have any update for this on your side, follow your comment here #39 ?
 
[image]

Thanks a lot,
Tuan

miangoleh, 2023-06-08:
@tuanho27 Can you share the code you've used?