autonomousvision/kitti360Scripts #39: FOV of Fisheye Cameras

niviruwijayaratne, 2021-11-07:
Hi, thank you for this great dataset!

I was wondering if you could possibly provide the vertical and horizontal FOV of the fisheye cameras...will be very helpful for dual fisheye -> equirectangular.  If not, do you have any recommendations on how the FOV can be estimated with the existing camera parameters or even better a recommended method for creating an equirectangular image? Thank you!

yiyiliao, 2021-11-11:
Hi, thanks for the question! The FOV is roughly 185 degrees in both directions. For the code of converting fisheye to equirectangular, I am checking with another colleague and will let you know if there is any update.

niviruwijayaratne, 2021-11-11:
Awesome thank you! Look forward to hearing from you

zcq15, 2022-09-20:
Hello, may I ask if the author will release the code for converting fisheye images to panoramic images, or is there a recommended third-party implementation tool? Looking forward to your reply. Thank you!

wangjksjtu, 2022-09-27:
Hi @yiyiliao, Thanks for the amazing works!! It would be fantastic if the utility functions for converting fisheye images to equirectangular images can be released. Is there an expected date for this? Looking forward to your reply!

yiyiliao, 2022-10-01:
Hi @zcq15 and @wangjksjtu , I haven't received feedback from my colleague for the conversion code yet, unfortunately. There are some public tools that might be helpful:
https://kaustubh-sadekar.github.io/OmniCV-Lib/Fisheye-to-equirectangular.html
http://paulbourke.net/dome/dualfish2sphere/