autonomousvision/kitti360Scripts #40: Some dynamic objects' 3D bbox labels are missing

songanz, 2021-11-08:
For example, in sequence "2013_05_28_drive_0000_sync", frame "0000000009", there is no 3D bbox label for the dynamic object (car) behind the vehicle. Please update the annotation. 
[image]
[image]

yiyiliao, 2021-11-10:
Thank you very much for pointing out the missing label! Indeed, the dynamic label of this car is missing. For this particular case, it is due to the fact that we ignored the first 250 frames of sequence "2013_05_28_drive_0000_sync" when converting 3D labels to 2D, i.e., our 2D semantic maps start from "0000000250" for this sequence. Therefore, we also recommend ignoring the first 250 frames when using 3D LiDAR scans.

songanz, 2021-11-15:
Thank you very much!

closed 2021-11-15 by songanz