autonomousvision/kitti360Scripts #42: cam0_to_world vs poses

mertkiray, 2021-11-19:
Hello, 
What is the difference between poses.txt and cam0_to_world.txt? 
 Also, what is the coordinate system of the poses?

Thanks.

yiyiliao, 2021-11-20:
Hi, you can find the description of the poses.txt on our documentation page:
Each line has 13 numbers, the first number is an integer denoting the frame index. The rest is a 3x4 matrix denoting the rigid body transform from _GPU/IMU coordinates_ to a world coordinate system. The origin of this world coordinate system is the same for all sequences, chosen as the center of the sequences. 

cam0_to_world instead provides the rigid body transform from the _rectified perspective camera_ coordinates to the world coordinate system.

Here is how cam0_to_world is calculated from poses.txt:
https://github.com/autonomousvision/kitti360Scripts/blob/7ecc14eab6fa30e5d2ac71ad37fed2bb4b0b8073/kitti360scripts/helpers/project.py#L36-L38

mertkiray, 2021-11-20:
Hello,
Thank you for your quick reply. 
I am trying to use the poses for LLFF(https://github.com/Fyusion/LLFF#using-your-own-poses-without-running-colmap) without running COLMAP. 

They say 
1. Make sure your poses are in camera-to-world format, not world-to-camera.
2. Make sure your rotation matrices have the columns in the correct coordinate frame [down, right, backwards].

So as I understand the poses in cam0_to_world.txt are camera to world format but the poses are in coordinate frame of 
forward, left, up so I have to change it to right up back. 

Should I use cam0_to_world.txt or poses.txt for this purpose?

Thanks a lot

mertkiray, 2021-11-20:
Also, it says that you generate a pose if the moving distance between this frame and the last valid frame is larger than a threshold. If not, can we assume that the pose is equal to the last generated pose? For example there is a pose for frame 300 for example but not for the 301th frame. Can we say the pose of frame 301 = pose of frame 300?

Thanks :)

yiyiliao, 2021-11-20:
> 1. Make sure your poses are in camera-to-world format, not world-to-camera.

Yes you can use cam0_to_world.txt for this purpose if you work with images collected from the left perspective camera (image_00). For other cameras, you need to calculate the camera-to-world transformation from poses.txt using the code I shared above.

> 2. Make sure your rotation matrices have the columns in the correct coordinate frame [down, right, backwards].

Yes, I think you need to rotate the rotation matrices as our columns are currently defined as [forward, left, up] as you understood.

>  Can we assume that the pose is equal to the last generated pose? 

No, the car is almost static but might move at a very slow speed. You can get an approximation via interpolation using the previous and the next available pose, but this may not be very accurate. Another way to get more accurate poses of those frames is to check our raw OXTS Measurements:
[image]

closed 2021-12-08 by mertkiray