autonomousvision/kitti360Scripts #43: Issues with world coordinate system

mertkiray, 2021-11-23:
Hello again.

I have debugged my approaches and tracked down the bug I have to the coordinate systems the 3D Velodyne points have and the camera poses. What I tried to see was to visualize the camera poses in a sequence and over impose the 3D Velodyne points on top of them to see how they would appear. The XYZ values of Velodyne points are in the range of -2 to +2 mostly while the translation part of my camera poses has values in the e+3 range. I will paste how I get the points and poses below so in case I am doing anything wrong. Sorry to bother you with questions like this but the whole system is quite difficult to understand just following the documentation. 

For pose calculation we used convertOxtsToPose function.  We use raw velodyne pointclouds for each frame. We assumed both Velodyne pcds and Poses are in world coords, which seems to be not correct. It would help us a lot if you could guide us.

[image]
Which the orange point is the visualized pose locaation

yiyiliao, 2021-11-23:
Hi, I am not sure whether I understood your goal. Do you want to overlay the camera trajectory on the Velodyne points?

I don't think you need convertOxtsToPose for this purpose. This is used to convert raw GPU/IMU measurements to the Euclidean coordinate frame. 

The raw Velodyne points are defined in the coordinate of the Velodyne instead of the world coordinate. To convert it to the world coordinate you need: velodyne -> cam0 (inversion of calib_cam_to_velo.txt); cam0 -> pose (from calib_cam_to_pose.txt).

mertkiray, 2021-11-23:
Thank you for your fast reply. We transformed the Velodyne points to world using the following:

`
TrCam0ToVelo = loadCalibrationRigid(self.fileCameraToVelo)
TrVeloToCam = np.linalg.inv(TrCam0ToVelo)

pcd = self.load_velodyne_data(frame)
pcd = np.matmul(pcd, TrVeloToCam)
pcd = np.matmul(pcd, TrCamToPose)
`

Then we take the camera pose in the world coords  from cam0_to_world.txt. When we visualize the Velodyne points and the camera pose in the world afterwards, we still get the same result above. I.e. the camera is not in the middle of the scanned 3D points but quite far away.

GreatGameDota, 2021-11-25:
You can align the vehicle and velodyne points by transforming them to the vehicle pose via velodyne -> cam0 (inversion of calib_cam_to_velo.txt); cam0 -> pose (from calib_cam_to_pose.txt); pose -> car_pose (from pose.txt).

An issue I'm having thats similar is knowing what coordinate is the car at, at a certain timestamp. We can get its rigid tranform at any given timestamp however are we able to get coordinates of where the car/camera is located from pose.txt or something else? Any help would be appreciated!

mertkiray, 2021-11-25:
@GreatGameDota thanks for the reply, do you know what coordinate system does the poses at cam0_to_world.txt use?

GreatGameDota, 2021-11-25:
I haven't used cam0_to_world for any transformations but based on the documentation it seems to convert 2d image coordinates to 3d world coordinates which isn't useful for point clouds. The velodyne points use the foward,left,up poses yes so you'd just need to convert them to right,up,back but negating the x,y points then swapping axises to y,z,x. I believe?

mertkiray, 2021-11-25:
I am asking because NeRF requires camera poses in right up back coordinate system. I think cam0_to_poses.txt gives us the camera poses but I could not understand their coordinate system to change to right up back.

mertkiray, 2021-11-25:
@yiyiliao is the poses in cam0_to_world.txt is in 
Perspective camera coordinate system: x = right, y = down, z = forward
or
World coordinate system: x = forward, y = left, z = up

yiyiliao, 2021-11-26:
> I am asking because NeRF requires camera poses in right up back coordinate system.

I don't think NeRF has any requirement on the world coordinate system. It says "the local camera coordinate system of an image is defined in a way that the X axis points to the right, the Y axis upwards, and the Z axis backwards as seen from the image." This defines the coordinate of pixel in the camera coordinate system [(see NeRF code)](https://github.com/bmild/nerf/blob/c39def3944c982aa7aabeacdd2e856c07da39b9e/run_nerf_helpers.py#L127):
`dirs = tf.stack([(i-W*.5)/focal, -(j-H*.5)/focal, -tf.ones_like(i)], -1)`

For our local camera coordinate frame defined as  x = right, y = down, z = forward, you need to adjust this line accordingly:
`dirs = tf.stack([(i-W*.5)/focal, (j-H*.5)/focal, tf.ones_like(i)], -1)`

These points in the camera coordinate frame are transformed to the world coordinate using the camear-to-world transformation matrx, c2w. 
`rays_d = tf.reduce_sum(dirs[..., np.newaxis, :] * c2w[:3, :3], -1)`
In KITTI-360,  `cam0_to_world.txt` is what you need for c2w if you use the left perspective camera (camera 0).

If you are interested in converting 3D points in the raw Velodyne scans to the world coordinate system, you need to convert them as @GreatGameDota explained:
`velodyne -> cam0 (inversion of calib_cam_to_velo.txt); cam0 -> pose (from calib_cam_to_pose.txt); pose -> car_pose (from pose.txt)`.

closed 2021-12-08 by mertkiray