autonomousvision/kitti360Scripts #44: Vehicle Coordinates

GreatGameDota, 2021-11-26:
How do we determine what coordinate is the car/vehicle at, at a certain timestamp? We can get its rigid transform (data_poses/pose.txt) at any given timestamp however are we able to get coordinates of where the car/camera is located from pose.txt or something else?

yiyiliao, 2021-11-26:
I am not sure if I understood your question correctly. If you are interested in getting where camera0 is located in the world coordinate frame, you can transform the origin (0,0,0) of the camera0's local coordinate frame to the world coordinate using `cam0_to_world.txt`. This means that the translation of `cam0_to_world.txt` tells where the camera0 is located. 

Similarly, 'poses.txt' tells where the GPS/IMU sensor is located in the world coordinate. For the other cameras, you can find where each camera is located in the local coordinate frame of the GPS/IMU sensor first (from `calib_cam_to_pose.txt`), and then transform them to the world coordinate using `poses.txt`.

GreatGameDota, 2021-11-26:
I have tried transforming [0,0,0,1] using cam0_to_world.txt however [0,0,0,1] x cam0_to_world just outputs [0,0,0].

yiyiliao, 2021-11-27:
If you are only interested in the location of the origin, you will have: 
`R x 0 + t = t`
where `R` is a 3x3 matrix, `0` and `t` are 3x1 vectors.
It means you can directly extract the translation vector from cam0_to_world.

GreatGameDota, 2021-11-27:
So its position in the world frame is the fourth column in the cam0_to_world.txt vector. How do I transform it to be useful though? I tried multiplying by the inverse of the whole matrix to transform it to cam0 but that didn't give a point in the center of the point cloud.

GreatGameDota, 2021-12-04:
I looked at it more and tried to use the last column of the poses.txt matrix but have the same issue where the point isnt anywhere near the point cloud points (which where projected to the car's pose using the same matrix).
[image]
Do I need to apply scaling to the point? Any help would be greatly appreciated!

yiyiliao, 2021-12-07:
Hi, it would be great if you can share a minimal test code for the above visualization. Thanks!

GreatGameDota, 2021-12-07:
Read in labeled/colored point cloud
```py
pcd = o3d.io.read_point_cloud(filename)
```
Project points to car_pose
```py
points = np.asarray(pcd.points)
points, center = project_points(points, timestamp)
```
Where the function project_points is:
```py
def project_points(points, timestamp):
    pcd_expanded = np.concatenate([points, np.ones((len(points),1))], axis=1)
    cam0 = pcd_expanded @ np.linalg.inv(cam_to_velo) # velo -> cam0
        
    pose = cam0 @ cam_to_pose # cam0 -> pose
    
    timestamp = min(pose_df.index.values, key=lambda x:abs(x-timestamp)) # Find closest pose timestamp

    pose_t = pose_df.loc[timestamp].values.reshape((3,4))
    pose_expanded = np.concatenate([pose_t, np.array([0,0,0,1]).reshape(1,4)])

    center = [pose_t[0][3], pose_t[1][3], pose_t[2][3]]
    car_pose = pose @ pose_expanded # pose -> car_pose

    return car_pose[:,:3], center
```
Where cam_to_velo is calib_cam_to_velo.txt, cam_to_pose is calib_cam_to_pose.txt, pose_df is poses.txt.
Add center point to projected points array:
```py
points = np.concatenate([points,[center]], axis=0)
colors = np.asarray(pcd.colors)
colors = np.concatenate([colors,[[0.5,0.5,0.5]]],axis=0)

pcd.points = o3d.utility.Vector3dVector(points)
pcd.colors = o3d.utility.Vector3dVector(colors)
```
Finally visualize all the points:
```py
o3d.visualization.draw_geometries([pcd])
```

Mohtasib, 2022-03-16:
Hi @yiyiliao and @GreatGameDota ,

Have you managed to solve this issue? Because I am struggling with the same problem.

I loaded the PLY files into Open3D and when trying to move the camera in Open3D to the cam0 poses from `cam0_to_world.txt` it just go to weired poses similar to what @GreatGameDota showed above.

eslambakr, 2022-04-18:
Hello All,

I think u have to use pose.txt. This is a simple example to get the ego vehicle position:
 
```
def _getPosMat(self, seqId):
        """
        Transform from GPU/IMU coordinates to a world coordinate system
        """
        pose_file = os.path.join(self.kitti360Path, 'data_poses', '2013_05_28_drive_%04d_sync' % seqId,
                                 'poses.txt')
        if not os.path.isfile(pose_file):
            raise ValueError('%s does not exist! \n'
                             'Please specify KITTI360_DATASET in your system path.\n'
                             'Please check if you have downloaded system poses (data_poses.zip)'
                             ' and unzipped them under KITTI360_DATASET' % pose_file)
        [ts, poses] = self._loadPoses(pose_file)
        poses = self._postprocessPoses(poses)
        frame_posmat_dict = {}  # key: frame id  & value: posMat
        for t, pose in zip(ts, poses):
            frame_posmat_dict[t] = pose
        return frame_posmat_dict
```
Followed by something like this:
```
frame_posmat_dict = self._getPosMat(seqId=seqID)
ego_pos = frame_posmat_dict[frameId][:3, 3]  # [x, y ,z] in the world coordinates.
```

I hope this helps and answer your question.

GreatGameDota, 2022-04-18:
I have tried that already:
```py
pose_df = pd.read_csv("KITTI-360/data_poses/2013_05_28_drive_0000_sync/poses.txt", sep=" ", header=None, index_col=0)

num = <some timestamp>
num = min(pose_df.index.values, key=lambda x:abs(x-num)) # Find closest pose timestamp
pose_t = pose_df.loc[num].values.reshape((3,4)) # I assume this is what your post process function does

pose_t[:3,3] # same as your code
```
This outputs `[844.2575,3731.067,115.3982]` which is the same point I display in the above image.

eslambakr, 2022-04-18:
I see. The seqID in your case is Zero.
Can u share the frameID with me to double check

GreatGameDota, 2022-04-18:
Timestamp/frameID 106

eslambakr, 2022-04-18:
I got the same values:
`ego_pos =  [ 844.2574508 3731.0665     115.3981776]`
Which means the issue in the visualization code I think as I am using the same approach in other application and it works fine.

GreatGameDota, 2022-04-19:
How are you projecting the point cloud points? My code for doing so is above, what could be wrong with it?

yiyiliao, 2022-04-20:
Sorry for the late reply. Let's discuss based on this simple 2D illustration:
[image]

As I understood, @GreatGameDota you were trying to transform the accumulated point cloud `p` from the *world coordinate* (with origin `o^w`) to the coordinate of the vehicle at one specific timestamp (with origin `o^c_t`), is that correct? In this case, you need:
```
car_pose = (np.linalg.inv(pose_expanded) @ pcd_expanded.T).T
```
Note that the origin is reset to `o^c_t` after this transformation.

Otherwise, you can also visualize the car center `o^c_t` and the points `p` without your `project_points` function, as both are defined in the world coordinate:
```
pcd = o3d.io.read_point_cloud(filename)
_, center = project_points(points, timestamp)

points = np.concatenate([points,[center]], axis=0)
colors = np.asarray(pcd.colors)
colors = np.concatenate([colors,[[0.5,0.5,0.5]]],axis=0)
```

GreatGameDota, 2022-04-20:
> ``` car_pose = (np.linalg.inv(pose_expanded) @ pcd_expanded.T).T ```

This worked thanks!

closed 2022-04-20 by GreatGameDota