autonomousvision/kitti360Scripts #46: Accumulaser poses

mertkiray, 2021-12-10:
Hello, can you please clarify what does system pose means in accumulaser output? 

What does 4x4 matrices in lidar_pose.dat mean?

lidar_loc.dat: Each row contains the system pose when the 3D point specified in the same row of lidar_points_{type}.dat is captured.
lidar_pose.dat: This contains the system poses from FIRST_FRAME to LAST_FRAME.

yiyiliao, 2021-12-15:
> When I project the accumulated pointcloud into a single frame image the invisible points in the accumated pointcloud are also visible in the image. So how can I get accumulated points visible in a certain camera frame?

This task is not trivial. In our 3D-to-2D label transfer method, we first reconstruct a mesh from the accumulated point cloud using Ball Pivoting, and then check whether a 3D point is visible or not at a given frame using this mesh. However, the reconstructed mesh is not very accurate in such outdoor scenes, thus limiting the accuracy of the visibility check. We can upload our C++ code for rendering accumulated point cloud to each frame if you think it would be useful.

> can you please clarify what does system pose means in accumulaser output?

lidar_loc.dat: Each row contains a transformation matrix that transforms a 3D point in its original sensor coordinate (Velodyne or SICK) to the world coordinate. The number of rows is the number of 3D points in lidar_points_{type}.dat.
lidar_pose.dat: Each row contains a transformation matrix that transforms a 3D point in the Velodyne coordinate to the world coordinate. The number of rows equals to the number of frames.

mertkiray, 2021-12-15:
> We can upload our C++ code for rendering accumulated point cloud to each frame if you think it would be useful.

This would be really helpful. Thanks

mertkiray, 2021-12-20:
Hey @yiyiliao, when can you share the C++ code.

Thank you so much :)

carloradice, 2022-02-03:
>  We can upload our C++ code for rendering accumulated point cloud to each frame if you think it would be useful.

There are any updates on this issue?
I also would like to get accumulated points visible in a certain camera frame.

yiyiliao, 2022-02-04:
Hi @mertkiray @carloradice, sorry for the late reply! I plan to clean up the code soon and release it next week.

mertkiray, 2022-02-25:
Hello @yiyiliao,
Any updates?

yiyiliao, 2022-02-27:
I am quite busy with the ECCV deadline and I am really sorry for the delay. As a temporary solution, you can find the uncleaned version here (including some test data and a test script):
https://drive.google.com/drive/folders/1Had3M1L4IZR4mN4X-bsy0P4rsgMJEIqs
I will clean up the code and commit it to our repo after the ECCV deadline.

closed 2022-03-24 by mertkiray

eslambakr, 2022-04-19:
Hello @mertkiray,
I wanna filter only the visible accumulated points in a certain camera frame.
Thus, Did u manage solve the issue using the aforementioned code? or using any other method?