autonomousvision/kitti360Scripts #48: AssertionError n_pts_loaded

DianaDI, 2022-01-12:
C:\Users\...\kitti360Scripts\venv\Scripts\python.exe C:/Users/.../kitti360Scripts/kitti360scripts/viewer/kitti360Viewer3D.py --sequence 0010
Loading C:\Users\...\DATA\data_3d_bboxes\train\2013_05_28_drive_0010_sync.xml...
...
Loaded 2669 instances
Loaded 14960 boxes
Found 17 ply files in 2013_05_28_drive_0010_sync
Loading C:\...\DATA\data_3d_semantics\train\2013_05_28_drive_0010_sync\static\0000000002_0000000208.ply 
Traceback (most recent call last):
  File "C:\...\kitti360Scripts\kitti360scripts\viewer\kitti360Viewer3D.py", line 313, in <module>
    pcd = v.loadWindow(pcdFile, args.mode)
  File "C:\...\kitti360Scripts\kitti360scripts\viewer\kitti360Viewer3D.py", line 178, in loadWindow
    data = self.annotation3DPly.readBinaryPly(pcdFile, n_pts)
  File "C:\...\kitti360Scripts\kitti360scripts\helpers\annotation.py", line 454, in readBinaryPly
    assert(n_pts_loaded==n_pts)
AssertionError

Process finished with exit code 1

I keep getting this error with original dataset loaded from the website. Could you please tell what might be the problem and how to solve it, if possible? I run it via pycharm.

Kind regards,
Diana

DianaDI, 2022-01-13:
If I remove this assert and just use n_pts instead of n_pts_loaded in that method, I have further problems in different part of the code. Moreover, `n_pts_loaded = len(plyData) / self.fmt_len ` is not an integer, not a whole number, that is strange. Could you please help me to resolve this? Maybe I miss something in configuration or something else...

DianaDI, 2022-01-13:
Might it be the case with different lib versions? could you please publish libs (including open3d) versions used in this repo? e.g. requirements.txt. Thanks!

Zhaozhpe, 2022-01-30:
> If I remove this assert and just use n_pts instead of n_pts_loaded in that method, I have further problems in different part of the code. Moreover, `n_pts_loaded = len(plyData) / self.fmt_len ` is [...]

Hello, I met the same problem. But I find that the proper value of `self.fmt_len` that satisfy `n_pts_loaded==n_pts` is 28. Therefore, I just change the value of `self.fmt` and `self.fmt_len` to 28 and '=fffBBBiiBi'. Then it works!

DianaDI, 2022-01-31:
@Zhaozhpe Thank you so much! It worked!

isunLt, 2022-03-12:
> 

You are right. According to the official document, every row of the ply file contains `x y z red green blue semanticID instanceID isVisible confidence` 10 words in total. However, `confidence` is missed in the code, which is a 32-bits float. Therefore, changing the value of `self.fmt` to 'fffBBBiiBf' and self.fmt_len to 28 can solve the problem.

yiyiliao, 2022-04-20:
Thank you very much @Zhaozhpe @isunLt for providing the correct answer! We released a new version of the ply file with `confidence` but didn't update the code in time. We have now switched to a more flexible [function](https://github.com/autonomousvision/kitti360Scripts/blob/b08d57b023d60c25d0baeb6b0e00d874284efadd/kitti360scripts/helpers/ply.py#L117) that can load any format of ply file without manually providing the data type. I will close this issue for now and feel free to reopen it if you have any further questions!

closed 2022-04-20 by yiyiliao