autonomousvision/kitti360Scripts #52: Empty dynamic point-cloud files. Accumulated point-clouds issue

eslambakr, 2022-04-06:
Dears,

I noticed that some dynamic point-cloud files are empty.
For example :

1) KITTI-360/data_3d_semantics/train/2013_05_28_drive_0000_sync/dynamic/0000000002_0000000385.ply
2) KITTI-360/data_3d_semantics/train/2013_05_28_drive_0000_sync/dynamic/0000000372_0000000610.ply
3) KITTI-360/data_3d_semantics/train/2013_05_28_drive_0000_sync/dynamic/0000000599_0000000846.ply

All of them has size of 311bytes.
I downloaded the file several times and each time the files are empty like that.

Thanks in advance.

yiyiliao, 2022-04-08:
Hi,

An empty dynamic point cloud file means that the corresponding window does not contain any dynamic object. These three files are also empty on our side but the next two are not:
285    000000002_000000385.ply
285    000000372_000000610.ply
285    000000599_000000846.ply
963K 000000834_000001286.ply
551K  000001270_000001549.ply
...

Best,
Yiyi

eslambakr, 2022-04-08:
Thank you very much for your prompt response.
It is clear now.
But does that mean the static files contain points for the static objects only? or the whole scene?

yiyiliao, 2022-04-11:
Yes, the static files contain points for the static objects only. You may combine both static and dynamic ply files to obtain the whole scene. You may also get the whole scene from the raw scans.

closed 2022-04-19 by eslambakr