autonomousvision/kitti360Scripts #54: [Bug] Visibility Flag is inverted in Sequence 0 for accumulated PCs

eslambakr, 2022-04-19:
Dears,

I have noticed that the visibility flag is inverted in Sequence 0, as shown below:
1) `showVisibleOnly=True`
[image]
2) `showVisibleOnly=False`
[image]

yiyiliao, 2022-04-20:
Hi, thank you for reporting the problem. This is actually not a bug. This part is not considered `visible` as we do not provide semantic labels for the starting frames of this sequence (0000). Therefore, the 3D semantic label is not retrieved from the corresponding 2D images in this region. Instead, we keep a 3D point only if it is uniquely labeled by a 3D bounding box and assign the label according to the annotation. 

If you check the next window (0000000372_0000000610.ply), the invisible points should be normal (black means points with `visible=0`):
[image]

eslambakr, 2022-04-27:
Thank you very much @yiyiliao .
I have verified that for the rest of frames in sequence zero.
Is this behavior should be followed for the rest of sequences?
I mean you do not provide semantic labels for the starting frames of each sequence or sequence zero only?

eslambakr, 2022-04-30:
Thank you very much for the clarification!

closed 2022-04-30 by eslambakr