autonomousvision/kitti360Scripts #55: Intrisic Camera Matrix

MarcusSchilling, 2022-04-19:
I wanted to ask if the intrinsic camera matrix is incorrect. Here is the intrinsic camera matrix loaded with the kitti360 scripts for P_rect_00:
array([[552.554261,   0.      , 682.049453,   0.      ],
       [  0.      , 552.554261, 238.769549,   0.      ],
       [  0.      ,   0.      ,   1.      ,   0.      ],
       [  0.      ,   0.      ,   0.      ,   1.      ]])
That means cx=682.049453 and cy= 238.769549. In my understanding the principle point should lay at the center of the image which would mean that: 2*cx should be the width of the image which is 1364,098906. But the Perspective Rectified image has a width of 1408. Same holds for height 2*cy=477,539098!=376=height. Am i wrong or is the calibration matrix not correct? 
I would appreciate an explanation.

Thank you in advance,
Marcus

yiyiliao, 2022-04-20:
Hi Marcus, the intrinsic camera matrix is correct. It is not necessary that `cx` and `cy` lay at the center of the image. I recommend this [blog](https://ksimek.github.io/2013/08/13/intrinsic/) for a more detailed explanation.

MarcusSchilling, 2022-04-30:
Thank you for the explanation!

closed 2022-04-30 by MarcusSchilling