autonomousvision/kitti360Scripts #61: Implementation of CRF

vobecant, 2022-05-28:
Dear authors, 

thank you very much for your paper and this codebase. Can I please ask you whether you provide the method for 3D->2D label mapping by CRF (section 4 of the paper) anywhere in this repository? If not, could you please add it?

Thank you very much in advance.

yiyiliao, 2022-06-14:
Hi, sorry for the late reply. We plan to release our CRF code in another repo after some cleaning up. I will let you know when it is released.

rollrollpig, 2023-02-28:
Hi, Can I please ask you whether you release the CRF code in another repo and where to find it?

Thanks.

yiyiliao, 2023-02-28:
Hi @rollrollpig , sorry that the code is not released yet. I will try to release it by April.