autonomousvision/kitti360Scripts #63: Intrinsic conversion

LoickCh, 2022-06-24:
How do you convert the intrinsic camera matrix given in pixels into meters?

yiyiliao, 2022-06-27:
You need to know the size of a pixel for converting pixels into meters, which is usually given in the sensor catalogue. We use a Flea 2 camera but don't remember exactly the type, unfortunately. Why are you interested in converting the intrinsic camera matrix given in pixels into meters?

LoickCh, 2022-06-27:
Hello, 

Thanks for the answer. I wanted to check that the camera's cone of vision could be seen by placing the image plane at a focal distance expressed in metres in a world reference.  In other words, I wanted to find the physical focal length, physical sensor size, etc. to place a camera in a world frame.