autonomousvision/kitti360Scripts #66: Missing Velodyne frames in SLAM Test set

jedeschaud, 2022-08-03:
There are many missing frames in the SLAM test set with Velodyne data: data_3d_test_slam.zip. For example:
 - Frames 3419, 3421, 3423... missing in test_1
 - Frames 55, 56, 58... missing in test_2
 - Frames 16779, 16781, 16783... missing in test_3

Could you add all the missing data to be able to test the SLAM?

yiyiliao, 2022-08-04:
Hi, we intentionally hold out these frames as they are ignored in the evaluation, i.e., we evaluate the pose of one frame if the moving distance between this frame and the last valid frame is larger than a threshold. This means some frames collected at a very low moving speed are neglected.

yiyiliao, 2022-08-12:
Sure. I will release the missing test frames in a separate folder soon and give you an update.