autonomousvision/kitti360Scripts #76: Unable to see any visualization in the UI

rreddy78, 2023-01-17:
I think I have all the data extracted properly but I dont see any visualization in UI. Output is as follows:

```
export KITTI360_DATASET=/opt/datasets/kitti360/KITTI-360/
sxv1kor@BMH7160115:/opt/datasets/kitti360/KITTI-360/kitti360Scripts/kitti360scripts/viewer$ ./kitti360Viewer.py 
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 568, resource id: 6295943, major code: 40 (TranslateCoords), minor code: 0
/opt/datasets/kitti360/KITTI-360/data_2d_semantics/train/2013_05_28_drive_0000_sync/instanceDict.json
Loading /opt/datasets/kitti360/KITTI-360/data_3d_bboxes/train/2013_05_28_drive_0000_sync.xml...
building                      :	 538
pole                          :	 356
traffic light                 :	 17
traffic sign                  :	 182
person                        :	 60
rider                         :	 18
car                           :	 901
truck                         :	 11
bus                           :	 1
caravan                       :	 1
trailer                       :	 25
train                         :	 0
motorcycle                    :	 22
bicycle                       :	 44
garage                        :	 288
stop                          :	 1
smallpole                     :	 94
lamp                          :	 117
trash bin                     :	 153
vending machine               :	 10
box                           :	 104
Loaded 2960 instances
Loaded 10994 boxes
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 685, resource id: 6296203, major code: 40 (TranslateCoords), minor code: 0
```

yiyiliao, 2023-01-18:
Hi, thank you for reporting the problem. Which PyQt version do you use?

rreddy78, 2023-01-24:
> Hi, thank you for reporting the problem. Which PyQt version do you use?

Sorry for delay in replying. The version is:        PyQt5                         5.14.1

Its installed with the command: 

sudo apt-get install python3-pyqt5
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pyqt5 is already the newest version (5.14.1+dfsg-3build1)

RolandGao, 2023-07-12:
Hi, I have the same problem