autonomousvision/kitti360Scripts #77: Potential wrong ground truth bounding boxes of pedestrians

zhongyidu, 2023-01-30:
Hi, I happened to find out some weird ground truth bboxes of pedestrians. For example, the pedestrians with globalID 24024 and 24025 in sequence 0 are:
- firstly labeled as static objects, which is not true since both of them are moving across the frames
- secondly have really huge length eg. 6m+ and unmatched bounding boxes.
Here are some visualizations of the specified pedestrian mapped onto the corresponding frame's pointcloud data. The green bbox is the ego vehicle and the black one is the pedestrian.
[image]
It's the frame 8476 of sequence 0 and the instance 24024 is clearly before the given bbox beside the car.
[image]
It's frame 8657 of sequence 0 and the instance 24025 hasn't moved into the bbox area and the bbox here basically includes nothing.

Across all the dataset, there're 8 instances of pedestrians which have the length bigger than 2m. I'm wondering if it's indeed mistakes in labeling and if so if there's any solution or workaround.

Thanks in advance:)

yiyiliao, 2023-02-03:
Hi, thank you for pointing out the inaccurate labels! 

At the very beginning of the annotation process, we used a large static bounding box to annotate a single dynamic object over all frames. This should have been replaced by annotating a dynamic object at every frame separately in the released version. Sorry for not cleaning all of them up.

I will fix these problems in the next update. A quick solution would be filtering out large, static pedestrian bounding boxes.

zhongyidu, 2023-02-08:
Hi, thanks for the answer! I will filter out the large static pedestrians for now as a workaround. There're indeed a couple of issues that I also found when I tried to explore the dataset:
- Logically the class riders and motorcycles/ bicycles should be paired which means for each frame if there's one rider, there should also be a matching motorcycle/ bicycle labeled in the frame. But the corresponding motorcycles/ bicycles seem to disappear for many frames across the entire dataset. I was trying to merge the riders with the corresponding motorcycles/ bicycles to create a similar class as cyclist in kitti dataset, which leads me to this question. I'm wondering if the objects are missing due to the labeling difficulty or sth else.
- I've noticed that sequence 2 started at frame 4391. I'd like to double-check if the first 4390 frames are not released for testing.
- And if possible, do you have any idea when the next update will be released:)

yiyiliao, 2023-02-09:
Thanks for pointing out additional issues!

> But the corresponding motorcycles/ bicycles seem to disappear for many frames across the entire dataset.

We annotated riders and motorcycles/ bicycles independently.  The motorcycles/bicycles sometimes disappear due to heavy occlusions, i.e., sometimes the motorcycles/bicycles are occluded but not the rider. Of course, it could also be that there is some noisy labeling that can be fixed.

> I've noticed that sequence 2 started at frame 4391. I'd like to double-check if the first 4390 frames are not released for testing.

Yes. They are not released for the purpose of testing.

> And if possible, do you have any idea when the next update will be released:)

Actually, I prefer to update this repo to filter out large static pedestrians if there are no other major issues. Updating the labels is possible but not trivial due to some accessibility issues.