autonomousvision/kitti360Scripts #79: How to transform the fisheye camera intrinsic param, include mirror_parameters, distortion_parameters and projection_parameters, into the OPENCV_FISHEYE parameter, such as k1,k2,k3,k4?

rockywind, 2023-02-21:
Hi,
I found the fisheye camera's parameter was different the [OPENCV_FISHEYE](https://docs.opencv.org/3.4/db/d58/group__calib3d__fisheye.html)'s k1,k2, k3, k4.
How to convert it to OPNECV_FISHEYE's parameters?
````
    def cam2image(self, points):
        """camera coordinate to image plane"""
        points = points.T
        norm = np.linalg.norm(points, axis=1)

        x = points[:, 0] / norm
        y = points[:, 1] / norm
        z = points[:, 2] / norm

        x /= z + self.fi["mirror_parameters"]["xi"]
        y /= z + self.fi["mirror_parameters"]["xi"]

        k1 = self.fi["distortion_parameters"]["k1"]
        k2 = self.fi["distortion_parameters"]["k2"]
        gamma1 = self.fi["projection_parameters"]["gamma1"]
        gamma2 = self.fi["projection_parameters"]["gamma2"]
        u0 = self.fi["projection_parameters"]["u0"]
        v0 = self.fi["projection_parameters"]["v0"]

        ro2 = x * x + y * y
        x *= 1 + k1 * ro2 + k2 * ro2 * ro2
        y *= 1 + k1 * ro2 + k2 * ro2 * ro2

        x = gamma1 * x + u0
        y = gamma2 * y + v0

        return x, y, norm * points[:, 2] / np.abs(points[:, 2])
````
The OPENCV_FISHEYE's distorted parameters.
[image]

yiyiliao, 2023-02-22:
Hi, I believe it cannot be simply converted as different camera models are used. We use the camera model proposed by Mei & Rives:
https://www.robots.ox.ac.uk/~cmei/articles/single_viewpoint_calib_mei_07.pdf

According to this [discussion](https://stackoverflow.com/questions/31089265/what-are-the-main-references-to-the-fish-eye-camera-model-in-opencv3-0-0dev), OpenCV uses the Kannala-Brandt camera model:
https://oulu3dvision.github.io/calibgeneric/Kannala_Brandt_calibration.pdf

If you want to use the Kannala-Brandt camera model, calibration is possible using our provided calibration sequences.

whu-lyh, 2023-04-06:
for Mei's fisheye model, I believe guys should refer to [[CamOdoCal](https://github.com/hengli/camera_model)](https://github.com/hengli/camera_model) for more details.

whu-lyh, 2023-04-09:
yes, i fixd the dead link~

closed 2023-04-13 by yiyiliao

sujitknandan, 2023-05-06:
Kitti 360 calibration also has p1 and p2 as additional distortion parameter apart from k1 and k2. But in the computation above those two parameters are not considered. Is there any specific reason or is the above calculation is outdated