autonomousvision/kitti360Scripts #84: Getting intensity in the segmented point cloud

wialb, 2023-03-23:
Hi,
The segmented point clouds for training and testing contain respectively the following properties:
    - [x, y, z, r, g, b, semantic, instance, visible, confidence];
    - [x, y, z, r, g, b, visible].

Is there any code to include the intensity into these point clouds?
I thought about changing some of the accumuLaser code to get the intensity alongside [x, y, z, r, g, b] and then work with knn to copy the intensity into the segmented point clouds. If there is an easier way, I would like to know!

Thanks in advance

yiyiliao, 2023-03-27:
Hi, changing the accumuLaser code should work, and sorry I don't think there is an easier way than that, unfortunately.

Shanci-Li, 2024-07-19:
Hi wialb! I am trying to extract the intensity of the KITTI360 in the segmented point cloud as well. Have you tried the method and does it work? If it is fine for you, I would appreciate it if you could share the point cloud with intensity or the script to do that!

Thanks!

wialb, 2024-08-25:
Hi Shanci-Li, sorry for the late answer. Unfortunately, I don't have the point clouds with intensity, nor the code. 
I wish you luck ;)