autonomousvision/kitti360Scripts #87: Fisheye camera & LiDAR calibration

HR-zju, 2023-04-12:
I'm glad you provided such a great dataset!
When I project the laser point cloud onto the fisheye camera image using kitti360Viewer3DRaw.py in the kitti360scripts tool, I find that there are significant offsets, such as the trees and vehicles in the image. What could this be causing?
[image]
[image]

HR-zju, 2023-04-12:
[image]

HR-zju, 2023-04-12:
@yiyiliao @pengsongyou @haofeixu @Pyrestone

yiyiliao, 2023-04-13:
Hi @HR-zju, thank you for reporting the issue! I am looking into it and will give you an update in 2-3 days.

yiyiliao, 2023-04-18:
Hi @HR-zju, I've found the problem. The raw velodyne points are not corrected for ego-motion. I have now added the correction code to the visualization script in this [commit](https://github.com/autonomousvision/kitti360Scripts/commit/478aeac239cffdf9858c9079c7d28cb45729362d). This is the result after fixing the problem:
[image]

Note that the ego-motion correction was considered in our data accumulation [code](https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/devkits/accumuLaser/src/accumulation.cpp#L450-L461), meaning that the accumulated point clouds are correct.

closed 2023-05-26 by yiyiliao

yuliangguo, 2023-08-08:
> Hi @HR-zju, I've found the problem. The raw velodyne points are not corrected for ego-motion. I have now added the ego-motion correction code to the visualization script in this [commit](https://github.com/autonomousvision/kitti360Scripts/commit/478aeac239cffdf9858c9079c7d28cb45729362d). This is the result after fixing the problem: [image] [image]
> 
> Note that the ego-motion correction was considered in our data accumulation [code](https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/devkits/accumuLaser/src/accumulation.cpp#L450-L461), meaning that the accumulated point clouds are correct.

Thanks for the correction. But there seems still certain offset along the car's border, and something gap on the trees, are they OK?  If I understand correct, some of the error may due to the lidar and came having different centers, thus different visible areas. Is there any way to handle such error?