autonomousvision/kitti360Scripts #89: 3d bounding boxes format mapping

psriram2, 2023-04-19:
Hi, 

Thank you for your great work. I am quite confused as to how the 3d bounding boxes are mapped to the individual frames. I'm attempting to plot all the 3d bounding boxes given a specific frame (image) from a sequence and I am not sure how to retrieve the 3d bounding boxes corresponding to that frame. Thanks for your help!

yiyiliao, 2023-04-24:
Hi, thank you for your interest in the dataset! You can retrieve the 3D bounding boxes of each frame based on the `semanticId` and `instanceId`. We provide `semanticId` and `instanceId` of each 3D bounding box in the `data_3d_bboxes`, this allows for retrieving the corresponding objects from the instance labels in the `data_2d_semantics`. We provide a script for this: 
https://github.com/autonomousvision/kitti360Scripts/blob/1e5d6371440c6c1b6d2e6d86d9a2869fe5ced45c/kitti360scripts/helpers/annotation.py#L318-L347

You can try it out via the [kitti360Viewer](https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/viewer/kitti360Viewer.py).
.

psriram2, 2023-04-24:
great, thank you! this worked for me - closing this issue

closed 2023-04-24 by psriram2