autonomousvision/kitti360Scripts #91: Mesh approximation for vegetations/trees

IoanaSil14, 2023-04-25:
Hi,

In the demo video it looks like the vegetation is approximated not by a bounding box but by a sphere-like object. Is there a way of obtaining these meshes?

Thank you.

[image]

IoanaSil14, 2023-04-28:
Hi,
thank you for the quick reply. If I understand correctly, the bounding boxes exist only for objects that have an instance and vegetation, for example, does not have a valid instance. Also, when loading the 2013_05_28_drive_0000_sync.xml file and counting the occurrences of vegetation I get none. 
How should I approach this problem?
Thank you for your help!

yiyiliao, 2023-05-10:
Hi, you are right that only a subset of classes are annotated with instance ID. In order to get count the occurrences of those classes without instance ID, the best solution is to project all bounding boxes within a certain distance to the current frame and keep it if it is not fully occluded. This is how we obtained the `visible_id` in [Panoptic NeRF](https://github.com/fuxiao0719/PanopticNeRF). You may also directly download the `visible_id` of a few sequences from there. 

Another more straightforward but less accurate solution is to select all bounding boxes whose distances to the camera are within a certain threshold (e.g., 100 meters) and those in front of the camera.