autonomousvision/kitti360Scripts #93: Project 3D bounding boxes on a fisheye image

wxn01, 2023-05-01:
Hi,
Thank you for your dataset!
In kitti360Viewer.py it is possible to project the 3d bounding boxes onto a perspective image. 
I would like to know if I change "CameraPerspective" to "CameraFisheye" in kitti360Viewer.py, can I project the 3d bounding boxes onto the fisheye picture? And is the bounding box projected on the fisheye picture also rectangular?
Thanks.

AnRanbel, 2023-05-26:
> Hi, Thank you for your dataset! In kitti360Viewer.py it is possible to project the 3d bounding boxes onto a perspective image. I would like to know if I change "CameraPerspective" to "CameraFisheye" in kitti360Viewer.py, can I project the 3d bounding boxes onto the fisheye picture? And is the 3d bounding box projected on the fisheye picture also cuboid? Thanks.

Have you solved it? I am also looking for 2d bboxof fisheye.

yiyiliao, 2023-05-26:
> Hi, Thank you for your dataset! In kitti360Viewer.py it is possible to project the 3d bounding boxes onto a perspective image. I would like to know if I change "CameraPerspective" to "CameraFisheye" [...]

It is possible to project the 3D bounding boxes on the fisheye picture, though our kitti360Viewer.py does not support this function yet. The edges of the 3D bounding boxes will not be straight anymore with the fisheye projection. In order to get the correctly projected 3D bounding boxes, it is required to sample points on the edges of the 3D bounding boxes (in addition to the 8 corners) and then project them.

wxn01, 2023-07-04:
Thank you for your reply. So are the 3d bounding boxes provided in the dataset now applicable to fisheye images?