autonomousvision/kitti360Scripts #95: Fisheye camera new distortion parameter

sujitknandan, 2023-05-06:
Kitti 360 calibration also has p1 and p2 as additional distortion parameter apart from k1 and k2. But in the computation above those two parameters are not considered. Is there any specific reason or is the above calculation outdated

yiyiliao, 2023-05-26:
Hi, we indeed do not consider the distortion parameters p1 and p2. The calculation is not outdated. I believe they are omitted for simplicity. I am not 100% sure though, as the fisheye calibration is done by my collaborator a long time ago.

miangoleh, 2023-06-08:
I was wondering what are the p1 and p2 as they are not defined in the MEI model. Can you provide more info or link me to where the calibration process is documented?

sujitknandan, 2023-06-09:
We got hold of the data from https://www.cvlibs.net/datasets/kitti-360/download.php.
Calibration data can be downloaded from https://s3.eu-central-1.amazonaws.com/avg-projects/KITTI-360/384509ed5413ccc81328cf8c55cc6af078b8c444/calibration.zip 

Attaching inline a sample 
%YAML:1.0
model_type: MEI
camera_name: image_02
image_width: 1400
image_height: 1400
mirror_parameters:
   xi: 2.2134047507854890e+00
distortion_parameters:
   k1: 1.6798235660113681e-02
   k2: 1.6548773243373522e+00
   p1: 4.2223943394772046e-04
   p2: 4.2462134260997584e-04
projection_parameters:
   gamma1: 1.3363220825849971e+03
   gamma2: 1.3357883350012958e+03
   u0: 7.1694323510126321e+02
   v0: 7.0576498308221585e+02
~

yiyiliao, 2023-06-09:
Hi @miangoleh , here you can find the [documentation](https://github.com/autonomousvision/kitti360Scripts/files/11702658/documentation.pdf). I believe that p1 and p2 refer to k3 and k4 in the documentation (which are not used in our projection model) but I am not 100% sure.

miangoleh, 2023-06-09:
@yiyiliao Thanks. The documentation is very helpful. The documentation is referring to some code as well. Can you also refer me to those as well?

yiyiliao, 2023-06-10:
@miangoleh Sorry I don't have the code. This repo should contain python implemenations of the same functions except for the `reconstruction` part. For example, the `project` function can be found [here](https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/helpers/project.py#L152).