autonomousvision/kitti360Scripts #96: velodyne data 3d raw ground truth poses

hj-joo, 2023-05-17:
Hi, I want to evaluate the performance using LiDAR SLAM. But I'm not sure what the ground truth for data_3d_raw is. The vehicle poses I received have a poses.txt, but the indexes do not match. For example, `2013_05_28_drive_0000_sync/velodyne_points/data` contains 11518 items of raw data, but `2013_05_28_drive_0000_sync/poses.txt` contains only 10514 information. This seems to mean that there is a lack of pose positions for each scan.

yiyiliao, 2023-05-17:
Hi, the frame indices may not be continuous, as `poses.txt` only provides the pose of a given frame if the moving distance between this frame and the last valid frame is larger than a threshold. This means some frames collected at a very low moving speed are neglected.

hj-joo, 2023-05-21:
If so, could you tell me how I can match the number of frames of 3d raw data and the index of poses.txt to the same number? Currently, I'm not sure how to measure the vehicle's pose only with LiDAR scans.

closed 2023-05-22 by hj-joo