autonomousvision/kitti360Scripts #97: is there have fisheye's bbox annotation?

AnRanbel, 2023-05-25:
I downloaded the fisheye dataset from the official website, but did not find the corresponding annotation, may I ask whether it is not provided yet? or am I not looking in the right place?

yiyiliao, 2023-05-26:
Hi, we do not provide annotation of the fisheye dataset currently. In our journal paper, we mention that we provide annotations of both perspective cameras. We are working on transferring labels to the fisheye images and there might be a release in the future.

AnRanbel, 2023-05-27:
> Hi, we do not provide annotation of the fisheye dataset currently. In our journal paper, we mention that we provide annotations of both perspective cameras. We are working on transferring labels to the fisheye images and there might be a release in the future. 

thank you for your reply.

closed 2023-05-27 by AnRanbel