autonomousvision/kitti360Scripts #103: Question about 3D bounding boxes

skmhrk1209, 2023-08-21:
Could you explain what `start_frame` and `end_frame` in the annotation files mean? 

> 3D annotations of objects containing instance labels. Each element object{d} denotes a bounding box having consistent semanticId and instanceId with the 2D labels. The vertices and faces matrices form the mesh of the bounding box in a local coordinate. The transform matrix transforms this mesh to the world coordinate. The timestamp denotes the frame ID for dynamic objects and is -1 for static object.

According to the above official description about 3D bounding boxes, for static objects, `start_frame` and `end_frame` can be interpreted as the first and last frames where the instance is projected, respectively. For dynamic objects, it is expected that `start_frame`, `end_frame`, and `timestamp` is the same, but it seems different.

yiyiliao, 2023-08-28:
Thank you for your question. For both static and dynamic objects, `start_frame` and `end_frame` denote how 3D bounding boxes are annotated, instead of the first and last frames where the instance is projected. More specifically, we split the dataset into windows of 200 meters for annotation, with each window distinguished by its `start_frame` and `end_frame`. A 3D bounding box is allocated in a window does not mean that it is visible in the full window.

skmhrk1209, 2023-08-28:
Thank you for your reply. I understood that each instance is not necessarily visible in the window defined by start_frame and end_frame. So, is it right that each dynamic object is projected to just one frame denoted by `timestamp`, each static object can be projected to any frames in the sequence, and the frames where the object is visible can be identified by checking whether the instance ID is included in the instance ID map of each frame?

yiyiliao, 2023-09-04:
Yes, you understand it correctly. Thank you for the further clarification.

skmhrk1209, 2023-09-04:
Thank you very much!

AmazingRoad, 2023-09-18:
> whether the instance ID is included in the instance ID map of each frame

hey man, could you tell me how to abtain the ID map?

skmhrk1209, 2023-09-19:
@AmazingRoad Check the official documentation.

closed 2023-09-19 by skmhrk1209