autonomousvision/kitti360Scripts #108: Ground Truth in Kitti-360 Vs Ground Truth in Kitti

Jesser1855, 2023-11-16:
Hello, 

I am working on multi-sensor fusion and 3d detection algorithms for my master thesis in TUM. 

In the normal kitti data-set [kitti data-se:](https://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d) : 

- image_2 
- velodyne, 
- calibration files 
-  label_2 (ground truth per frame) are provided. 

The issue is that I need a dataset (image and lidar data) taken in sequence and not like the ones provided by the normal kitti dataset. 
I found out that the kitti-360 is the kitti dataset that I need. 

**MY question here,** does the kitti-360 provides calibration and ground truth like the ones provided in normal kitti dataset ?

**this is the format of the label_2 (ground truth per frame )files  in the normal kitti [kitti ground truth as label_2**.](https://github.com/bostondiditeam/kitti/blob/master/resources/devkit_object/readme.txt)
-------------------------------

#Values    Name      Description
----------------------------------------------------------------------------
   1    type         Describes the type of object: 'Car', 'Van', 'Truck',
                     'Pedestrian', 'Person_sitting', 'Cyclist', 'Tram',
                     'Misc' or 'DontCare'
   1    truncated    Float from 0 (non-truncated) to 1 (truncated), where
                     truncated refers to the object leaving image boundaries
   1    occluded     Integer (0,1,2,3) indicating occlusion state:
                     0 = fully visible, 1 = partly occluded
                     2 = largely occluded, 3 = unknown
   1    alpha        Observation angle of object, ranging [-pi..pi]
   4    bbox         2D bounding box of object in the image (0-based index):
                     contains left, top, right, bottom pixel coordinates
   3    dimensions   3D object dimensions: height, width, length (in meters)
   3    location     3D object location x,y,z in camera coordinates (in meters)
   1    rotation_y   Rotation ry around Y-axis in camera coordinates [-pi..pi]
   1    score        Only for results: Float, indicating confidence in
                     detection, needed for p/r curves, higher is better.
                     
                     
   --------------------------------------------------------------------------------------          
   
           **If it does not deliver such label_2 files directly, Is there a way to build such files from other files provided by the kitti-360 ?** 
           
           Thank you in advance 
           best regards 
           Jesser Bouzid

JohnalDsouza, 2024-07-11:
Im facing the same problem. the documentation of kitti360 for 3d bounding boxes isnt very good. I am trying to visualize the 3d bounding boxes in BEV along with the lidar points. but due too lack of documentation, its a real struggle

yiyiliao, 2024-07-11:
Hi, one student of our team is working on providing a script for this. We will add it to the devkit once it is ready.

abhi1kumar, 2024-07-12:
Hi @JohnalDsouza 
You could check out the converted KITTI-360 in KITTI labels in our [SeaBird repo](https://github.com/abhi1kumar/SeaBird/tree/main/PanopticBEV) KITTI-360 section. It contains processed files in the KITTI format.

NaveenPrasaad, 2024-08-13:
> Hi, one student of our team is working on providing a script for this. We will add it to the devkit once it is ready.

Hello @yiyiliao, We are working on a model that takes KITTI format for BEV generation. Is the script for converting KITTI 360 to KITTI dataset available so that we can convert and use it for our model?

Hello @abhi1kumar, Actually we tried using the processed files in the KITTI format from SeaBird repo, but as our model needs the velo_to_cam coordinates in the calibration file, we got an error and hence we couldn't use it. Additionally we need road labels in label file to get the drivable area which is also not available in the SeaBird repo.

HarisAmir, 2024-09-18:
> Hi, one student of our team is working on providing a script for this. We will add it to the devkit once it is ready.

Hello @yiyiliao 👋

Any update on this end? I was also interested in converting the labels from **KITTI-360** to **KITTI** format

Petros626, 2025-08-12:
@yiyiliao Quite some time has passed and I'd like to ask about the progress.

yiyiliao, 2025-08-14:
Sorry for the delay. I have pushed it here: https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/conversion/convertKITTIBbox.py

Note that in KITTI-360, the camera is tilted downward by about 5° relative to the ground plane. As a result, a rotation matrix constructed from the yaw angle alone will also include this tilt, rather than being perfectly parallel to the ground. Here is an example:
[image]

In addition, as I am not fully familiar with KITTI’s bounding box definition, there may be inconsistencies in the conversion. Any feedback or reports of such inconsistencies would be greatly appreciated.

Petros626, 2025-08-14:
@yiyiliao when you project in KITTI360 the 3D BB to a point cloud or image, then they're parallel to the ground? Only when you convert them to KITTI they're tilted downwards?

The KITTI format for the 3D BB are: 
`3 dimensions 3D object dimensions: height, width, length (in meters)`
`3 location   3D object location x,y,z in camera coordinates (in meters)`
`1 rotation_y Rotation ry around Y-axis in camera coordinates [-pi..pi]`

Petros626, 2025-08-14:
I see the difference now, the question is how to align the both (w. 5° downwards, w/o 5° downwards) coordinate systems?

The conversion code you provide doesn't do this properly?

Petros626, 2026-09-02:
@yiyiliao Hey, 
Can you refer to that to close the discussion?