autonomousvision/kitti360Scripts #111: Does not support pillow versions > 9.0.0

AniketDhar, 2024-01-10:
I am trying to run kitti360Viewer3DRaw.py with Python 3.8 and pillow 9.3.0 but always gives me error 
`Please install the module 'Pillow' for image processing, e.g.
pip install pillow`
I have checked my python, pip and pillow paths and installations and everything seems to be alright.
I see that the following line https://github.com/autonomousvision/kitti360Scripts/blob/7d44b19446b92801cff403ac0eff5985986ff481/kitti360scripts/helpers/csHelpers.py#L19 checks for `PILLOW_VERSION` which has been discontinued from Pillow 9.0.0. 
The recommended way to use seems to use `__version`, for example,
`from PIL import __version__ as PILLOW_VERSION`

wxyLuna, 2024-05-22:
Hi, did you resolve this problem in the end? I tried to replace line 19 with 'from PIL import __version__ as PILLOW_VERSION' but same thing still happened  to me.
'''
Please install the module 'Pillow' for image processing, e.g.
pip install pillow

'''

jxbb233, 2024-07-20:
@wxyLuna You can try running `pip install -e .` after making the modifications.