autonomousvision/kitti360Scripts #112: code for PCL baseline

Linusnie, 2024-01-20:
Hi, is the code for the point cloud based image synthesis baseline (from [here](https://www.cvlibs.net/datasets/kitti-360/eval_nvs_rgb_detail.php?benchmark=nvs_rgb&result=f279b9aa8303006397a9a986fe1a8d992b486e1d)) available anywhere?

yiyiliao, 2024-02-04:
Hi, sorry that this code is not published as it is not well-documented. In case you don't mind, I can share the private repo with you.

Linusnie, 2024-02-04:
@yiyiliao that would be great! I'm looking into implementing something similar-ish, so would be helpful to have it as a starting point

yiyiliao, 2024-02-15:
Hi @Linusnie, I sent you an invitation. Feel free to contact me vial email if you'd like to know more about the code.