autonomousvision/kitti360Scripts #115: Some problems about the fisheye camera intrinsic and extrinsic matrix

Zanghu-ze, 2024-02-14:
Hello! I think you guys give us a wonderful dataset. But when I try to use the scripts to project the point on the fisheye image. I found there are some projection errors.  You can see that there are some gaps in the middle of the image and some shifts in the border origin. And when I tried to undistorted the fisheye image I found the same questions. So I have a question about the correctness of the intrinsic and extrinsic matrix. Can you check it out?
[image]

yiyiliao, 2024-02-15:
Hi, thank you for your interest in our dataset! The projection should be correct if you pulled the latest version (after adding ego-motion correction as discussed in this [issue](https://github.com/autonomousvision/kitti360Scripts/issues/87)). Of course, the intrinsic and extrinsic parameters are not 100% accurate but the noise should be minor. The error observed in the border is mainly because the camera centers are located differently from the lidar center.

Zanghu-ze, 2024-02-17:
@yiyiliao Thanks for you suggestion and comment! I will check these problems later!

closed 2024-02-17 by Zanghu-ze