autonomousvision/kitti360Scripts #116: Error CYTHONIZE_EVAL= python setup.py build_ext --inplace

Lee-JaeWon, 2024-03-18:
I configured that repository with Python 3.6 and followed the instructions in the README. In addition, I got the following error

```
CSUPPORT is required for unwrapping the velodyne data!
Run ``CYTHONIZE_EVAL= python setup.py build_ext --inplace`` to build with cython
```

I followed the above command and it seems to build, but I keep getting that error and it won't import.

jxbb233, 2024-07-20:
I had a similar issue and found a solution. Navigate to your local project directory and run `pip install -e .` after executing the command you mentioned. This should let you import the `curlVelodyneData` module without any issues.

Lee-JaeWon, 2024-07-20:
@jxbb233 

Thanks for the answer!

I solved it in the following way.

1. `CYTHONIZE_EVAL=1 python setup.py build_ext --inplace`
2. `python setup.py install`