autonomousvision/kitti360Scripts #118: Inquiry about Class Proportions in the Kitti360 Dataset

mona4399, 2024-04-05:
Dear authors,

I'm working with the Kitti360 dataset and seeking the class proportions within it. I couldn't find this in the documentation or issues. Could anyone share these details or direct me to where I might find them? Thanks for your help.

Best

yiyiliao, 2024-04-07:
Hi, we provide dataset statistics in our supplementary:
[image]

Let me know if this is not what you are looking for.

mona4399, 2024-04-07:
Thanks a lot,  that's exactly what I needed. Could you also provide the proportions for the unlabeled classes that will be ignored during evaluation? Much appreciated.