autonomousvision/kitti360Scripts #119: fisheye depth using dist or z-buffer

yuliangguo, 2024-04-06:
https://github.com/autonomousvision/kitti360Scripts/blob/9defe218f4d05a1c920924f38ebfa8efc27318da/kitti360scripts/helpers/project.py#L200C22-L201C1

I found that the depth preparation for perspective camera uses z-buffer however the preparation for fisheye using dist. Is there any particular reason for these inconsistant ways?

yiyiliao, 2024-04-07:
Hi, the reason is that the fisheye camera model is not a normal perspective camera model. The points are first projected to a unit sphere, see this
[documentation](https://github.com/autonomousvision/kitti360Scripts/files/14895571/documentation.pdf).

yuliangguo, 2024-04-07:
Good to confirm, thx for the reference.

closed 2024-04-07 by yuliangguo

wpfhtl, 2024-04-07:
Hi, is there any code to project the pcd file in world frame to the cameras