autonomousvision/kitti360Scripts #123: how does the fisheye camera trigered by lidar?

ZOUYIyi, 2024-06-21:
in the nuscence dataset , the camera is triggered when the lidar sweeps the center of the camera FOV.
is it the same case for the fisheye camera in this dataset ?
but the fisheye camera has big FOV, so even it is tiggered like this ,  the lidar points which projected to the left part of image  have big timestamp diffrence than the lidar points which projected to the right part of image.
how to solve this ? 
or it just interpolation the moving obj box ,so dont consider about the points projected to the image ?

ZOUYIyi, 2024-07-12:
> All the cameras are triggered at the same time when the laser faces forward.
> 
> When projecting lidar points to the cameras, you need to unwarp the LiDAR points to compensate for ego-motion. You can find scripts here: https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/viewer/kitti360Viewer3DRaw.py#L132-L139

for static objects ,it works fine ,but for the moving objects , the compensate cant work , the points in the moving object cant project to the right pixel of the image , how to solve this problem?

ZOUYIyi, 2024-07-12:
anyone have the project image  to check if the moving object points cant project right?

yiyiliao, 2024-07-12:
Indeed, for dynamic objects, there is no trivial solution to solve this problem..