autonomousvision/kitti360Scripts #124: Is there any method to convert 2D points into 3D?

adelatatar, 2024-08-10:


yiyiliao, 2024-08-16:
Could you elaborate more on your question? What kind of 2D points do you like to convert into 3D?

adelatatar, 2024-08-16:
Hi, thanks for your answer. 
Yes, I will elaborate. I want to perform unsupervised segmentation on the 3D dataset. For this I have an approach that uses 2D dataset. I perform a pre-segmentation on the 3D dataset and I use your scripts to convert each 3D segment into the corresponding 2D image (I use only the points from perspective cameras). After applying some segmentation models on the 2D patches (obtained from the 3D points mapped in the images), I want to convert them back into corresponding 3D points. Of course, I can keep the mapping between each 3D point and the corresponding 2D point when I perform the projection in the first place, but I was wondering if this "back-projection" from 2D points into 3D is implemented too. 

By the way, thank you very much for the provided scripts!!

yiyiliao, 2024-08-17:
Thank you for elaborating on the question. We don't have this back-projection implemented in our current script. It should be easy to implement with the provided intrinsic and extrinsic parameters. You may also find some existing libraries helpful, such as https://www.open3d.org/docs/0.7.0/python_api/open3d.geometry.create_point_cloud_from_depth_image.html

closed 2024-09-02 by adelatatar