autonomousvision/kitti360Scripts #125: Inquiry about SICK laser intensity

Shanci-Li, 2024-08-15:
Hi Yiyi! Thanks for your great work and effort in maintaining the repo. I am currently trying to generate the accumulated point clouds with intensity from raw 3D data. For Velodyne, I know the intensity is saved in the 4th column in the binary file. But it seems there are only 2 columns for SICK scans. Does that mean there is no intensity provided with SICK scans?

Best regards,
Shanci

yiyiliao, 2024-08-16:
Hi Shanci, thank you for your interest! Yes, our provided data do not include the intensity of SICK scans.

Shanci-Li, 2024-08-19:
Hi Yiyi! Thanks for your help! Besides the SICK intensity, I have a question regarding an explanation about Velodyne intensity you had in another issue #73 : 'Also, values above 100 indicate retro-reflective targets.'. I checked the distribution of the intensity in the accumulated point cloud I had recovered, the normalized intensity mostly ranges from 0-0.4 (which is 0-100 for 8-bit scale). To generalize the dataset with the high-accuracy lidar scans (intensity in 16 bits), do you think it is a good idea to cut off the intensity above 0.4 and normalize the intensity again?