autonomousvision/kitti360Scripts #126: How to assign the bounding boxes of static objects to frames?

CesarLiu, 2024-08-19:
Dear @yiyiliao ,

thanks a lot for the great work first.

For the 3d bbox annotation, the static objects are only labeled once, but one single static object(unique global instance) can be visible in a lot of time frames. Are there approaches to assign these static objects to all the frames they appearred?
The Instance labels from data_2d_semamtics can be used to check if the instance are in the perspective cameras, which may not be sufficient to get all instances in the current time frame, is it right?

Thanks a lot in advance!

Best

yiyiliao, 2024-08-23:
Thank you for your interest in our dataset!

For bounding boxes of static objects at a single frame, do you refer to *instance* bounding boxes or *all* bounding boxes including instance and stuff? If you are only interested in the *instance* bounding boxes, it should be a good way to get all instances using the data_2d_semantics, except for some far-away objects that are missed in the 2D labels.

CesarLiu, 2024-08-23:
Dear Prof. Liao,

thanks a lot for the quick reply. I do only care about the instance. It is convenient to extract from data_2d_semantics, but it covers only the front view. How about the area, which is not covered by the perspective cameras?

Thanks again for your attention!
Best